TeamWalrus/Walrus #86: Feature Request

sho-luv, 2018-04-02:
Is there a way to manually write a card. Lets say I have reader and I pull card data without bluetooth  or I get it some other way. Then I want to write it to a card using Walrus. If in the settings there was a manual enter card data that would be useful. So I could manually enter it an then write or emulate a card with walrus.

megabug, 2018-04-03:
Yep, manual card data entry is on our to-do list. This will probably take the form of a "virtual device" which redirects all card read requests to a pop-up dialog box where the user can manually enter card data (i.e. Wiegand data, UID, etc.)

I'll move it up our priority list - it's a relatively simple feature!

megabug, 2018-04-21:
The generic infrastructure for this and its implementation for `HIDCardData` has been added in 39654de2be872dd696df10a66e865d3bee33fb98.

The implementation for `ISO14443ACardData` will come after its tidy-up.