TeamWalrus/Walrus #97: No Card Devices Connected

h4waii, 2018-05-25:
Not sure if this is a feature request, or a bug report.

I seem to be having an issue with Walrus detecting my Proxmark as being connected..
I know it's connected, I can access it via Termux on /dev/ttyACM0 and use the "stock" proxmark3 client. I can see it show up in dmesg too.

I get the prompt to allow Walrus access to the USB device AFTER I unplug the proxmark3 or OTG cable.

Is there any way to attach Walrus directly to a specified port? An option would be to include a request for root access in order to interface with the port directly.

Hopefully this can be added!

jattilioz, 2018-06-06:
Similar experience, I get the same prompt after I unplug the Proxmark. Android 7.0

h4waii, 2018-06-07:
For what it's worth, I've finally updated to LineageOS 15.1 which is Android 8.1 Oreo, and this is no longer an issue. Device is detected after a few seconds of being plugged in, and is accessible.

megabug, 2018-06-15:
This is definitely a bug report; after connection, devices should be detected by Android and, after permission is granted, seen by Walrus as well.

Am I understanding correctly that this is/was happening:

1. Start Walrus
2. Connect USB device
3. Nothing happens
4. Disconnect USB device
5. Asked to grant permission, grant permission
6. Reconnect USB device
7. Device is now seen by Walrus

...or is it something else?

Either way, I'm afraid that this is probably something that will have to be diagnosed through better debug logging which isn't present in the current 0.1 version. The next version will have more verbose logging so we can diagnose these sorts of issues, so please give that a try when it's released (I'll try to remember to reply here with instructions once it's out).

Also, @h4waii: Do you know what OS version (and underlying Android version) you were using before your upgrade?

h4waii, 2018-06-15:
1. Start Walrus
2. Connect USB device
3. Nothing happens
4. Disconnect USB device
5. Asked to grant permission, grant permission
6. Reconnect USB device
7. Device is NOT seen by Walrus

I was using LineageOS 14.1, which is based on Android 7.1

audibleblink, 2018-06-22:
Identical experience on Android P RC2.
Permission is asked after the device is disconnected.

MeulemanSven, 2018-07-26:
Having a similar issue, but first a question. Are you guys running the release version or a self compiled version? Running the version from the playstore I have no issues. But wanting to take advantage of the Mifare option I compiled both the master and dans-branch. For both branches was a signed APK generated, the application runs fine but the proxmark3 is not found. I've got a proxmark3 rdv4. Going between the release and the master I noticed that the vendorid and productid changed from decimal to hex. I added the decimals but no change. Can anyone help getting in the direction of a solution.
My platform is a Oneplus 5 running Oreo 8.1.0.

realizedf, 2018-08-10:
if you are using iceman's fork or another fork then this will happen - flash the official firmware and should be good - if there is not enough power to the proxmark either get a cable that can supply power or turn on the proxmark w/ battery attached also.

had this problem with walrus + andprox etc with iceman's fork but with official firmware everything is good.

msoose, 2020-02-28:
Seems to be resolved now.

closed 2020-02-28 by msoose