TeamWalrus/Walrus #98: "Must specify preferenceTheme in theme" when attempting to view settings

ss23, 2018-06-16:
On LineageOS 14, I recieve the following backtrace when attempting to access the settings (after I click the menu button and tap "Settings"):
```
06-17 06:50:38.720 7176-7176/com.bugfuzz.android.projectwalrus.debug E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.bugfuzz.android.projectwalrus.debug, PID: 7176
    java.lang.RuntimeException: Unable to start activity ComponentInfo{com.bugfuzz.android.projectwalrus.debug/com.bugfuzz.android.projectwalrus.ui.SettingsActivity}: java.lang.IllegalStateException: Must specify preferenceTheme in theme
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2684)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2751)
        at android.app.ActivityThread.-wrap12(ActivityThread.java)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1496)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:154)
        at android.app.ActivityThread.main(ActivityThread.java:6186)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:889)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:779)
     Caused by: java.lang.IllegalStateException: Must specify preferenceTheme in theme
        at android.support.v7.preference.PreferenceFragmentCompat.onCreate(PreferenceFragmentCompat.java:211)
        at android.support.v4.app.Fragment.performCreate(Fragment.java:2331)
        at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1386)
        at android.support.v4.app.FragmentTransition.addToFirstInLastOut(FragmentTransition.java:1188)
        at android.support.v4.app.FragmentTransition.calculateFragments(FragmentTransition.java:1071)
        at android.support.v4.app.FragmentTransition.startTransitions(FragmentTransition.java:115)
        at android.support.v4.app.FragmentManagerImpl.executeOpsTogether(FragmentManager.java:2380)
        at android.support.v4.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManager.java:2338)
        at android.support.v4.app.FragmentManagerImpl.execPendingActions(FragmentManager.java:2245)
        at android.support.v4.app.FragmentManagerImpl.dispatchStateChange(FragmentManager.java:3248)
        at android.support.v4.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManager.java:3200)
        at android.support.v4.app.FragmentController.dispatchActivityCreated(FragmentController.java:195)
        at android.support.v4.app.FragmentActivity.onStart(FragmentActivity.java:597)
        at android.support.v7.app.AppCompatActivity.onStart(AppCompatActivity.java:177)
        at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1249)
        at android.app.Activity.performStart(Activity.java:6701)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2647)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2751) 
        at android.app.ActivityThread.-wrap12(ActivityThread.java) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1496) 
        at android.os.Handler.dispatchMessage(Handler.java:102) 
        at android.os.Looper.loop(Looper.java:154) 
        at android.app.ActivityThread.main(ActivityThread.java:6186) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:889) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:779)
```

dunderhay, 2018-06-19:
Thanks @ss23, I have run into this bug before, but only on recent builds as a result of moving to the appcompact Support Libraries. I wasn't running into the issue while using the build from the app store, could you just confirm if you are, or have compiled from GitHub? 

The cause of the issue is the settings activity has no theme preference set. This can be fixed by adding the line <item name="preferenceTheme">@style/PreferenceThemeOverlay.v14.Material</item> in the styles.xml file to specify an application theme to use. 

styles.xml with updates:
```
<resources>

    <style name="App" parent="Theme.AppCompat.Light.NoActionBar">
        <item name="colorPrimary">@color/primaryColor</item>
        <item name="colorPrimaryDark">@color/primaryDarkColor</item>
        <item name="colorAccent">@color/secondaryColor</item>
        <item name="preferenceTheme">@style/PreferenceThemeOverlay.v14.Material</item>
    </style>

    <style name="App.WithTransitions"/>

    <style name="App.WithTransitions.LauncherActivity">
        <item name="android:windowBackground">@color/primaryDarkColor</item>
    </style>

</resources>
```

You will also need to add the preferenceTheme dependency to build.gradle:
implementation 'com.android.support:preference-v14:27.1.1'

I think there may be another bug after you patch that one, please let me know :) If you are building off current master, I think there are changes like mentioned earlier that have essentially broken a lot of the preferences that were working (thanks @megabug) 😿

ss23, 2018-06-19:
@dunderhay Yes, this is compiled from master. Happy to try out those changes you mentioned. Presume you'd like a PR if they help?

dunderhay, 2018-07-07:
I have commited a fix for the issue reported ([f8efc3a96b0fa64feec9b924d8ae0c7ccbb4f04b](https://github.com/TeamWalrus/Walrus/commit/f8efc3a96b0fa64feec9b924d8ae0c7ccbb4f04b)) on my branch, but it doesn't resolve opening the settings activity yet. Another error as result of moving to support library. Will pull into master once all underlying issues related to settings activity are resolved. 

Debug output for new issue:

```
7-07 22:05:57.131 27782-27782/com.bugfuzz.android.projectwalrus.debug E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.bugfuzz.android.projectwalrus.debug, PID: 27782
    java.lang.RuntimeException: Unable to start activity ComponentInfo{com.bugfuzz.android.projectwalrus.debug/com.bugfuzz.android.projectwalrus.ui.SettingsActivity}: android.view.InflateException: Binary XML file line #43: Error inflating class com.bugfuzz.android.projectwalrus.card.ui.DeleteAllCardsPreference
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2778)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2856)
        at android.app.ActivityThread.-wrap11(Unknown Source:0)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1589)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6494)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
     Caused by: android.view.InflateException: Binary XML file line #43: Error inflating class com.bugfuzz.android.projectwalrus.card.ui.DeleteAllCardsPreference
        at android.support.v7.preference.PreferenceInflater.createItem(PreferenceInflater.java:258)
        at android.support.v7.preference.PreferenceInflater.createItemFromTag(PreferenceInflater.java:287)
        at android.support.v7.preference.PreferenceInflater.rInflate(PreferenceInflater.java:354)
        at android.support.v7.preference.PreferenceInflater.rInflate(PreferenceInflater.java:356)
        at android.support.v7.preference.PreferenceInflater.inflate(PreferenceInflater.java:162)
        at android.support.v7.preference.PreferenceInflater.inflate(PreferenceInflater.java:112)
        at android.support.v7.preference.PreferenceManager.inflateFromResource(PreferenceManager.java:137)
        at android.support.v7.preference.PreferenceFragmentCompat.addPreferencesFromResource(PreferenceFragmentCompat.java:432)
        at com.bugfuzz.android.projectwalrus.ui.SettingsActivity$SettingsFragment.onCreatePreferences(SettingsActivity.java:43)
        at android.support.v7.preference.PreferenceFragmentCompat.onCreate(PreferenceFragmentCompat.java:224)
        at android.support.v4.app.Fragment.performCreate(Fragment.java:2331)
        at android.support.v4.app.FragmentManagerImpl.moveToState(FragmentManager.java:1386)
        at android.support.v4.app.FragmentTransition.addToFirstInLastOut(FragmentTransition.java:1188)
        at android.support.v4.app.FragmentTransition.calculateFragments(FragmentTransition.java:1071)
        at android.support.v4.app.FragmentTransition.startTransitions(FragmentTransition.java:115)
        at android.support.v4.app.FragmentManagerImpl.executeOpsTogether(FragmentManager.java:2380)
        at android.support.v4.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManager.java:2338)
        at android.support.v4.app.FragmentManagerImpl.execPendingActions(FragmentManager.java:2245)
        at android.support.v4.app.FragmentManagerImpl.dispatchStateChange(FragmentManager.java:3248)
        at android.support.v4.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManager.java:3200)
        at android.support.v4.app.FragmentController.dispatchActivityCreated(FragmentController.java:195)
        at android.support.v4.app.FragmentActivity.onStart(FragmentActivity.java:597)
        at android.support.v7.app.AppCompatActivity.onStart(AppCompatActivity.java:177)
        at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1334)
        at android.app.Activity.performStart(Activity.java:7029)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2741)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2856) 
        at android.app.ActivityThread.-wrap11(Unknown Source:0) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1589) 
        at android.os.Handler.dispatchMessage(Handler.java:106) 
        at android.os.Looper.loop(Looper.java:164) 
        at android.app.ActivityThread.main(ActivityThread.java:6494) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807) 
     Caused by: java.lang.ClassCastException: com.bugfuzz.android.projectwalrus.card.ui.DeleteAllCardsPreference cannot be cast to android.support.v7.preference.Preference
        at android.support.v7.preference.PreferenceInflater.createItem(PreferenceInflater.java:251)
        at android.support.v7.preference.PreferenceInflater.createItemFromTag(PreferenceInflater.java:287) 
[... cut ...]
        at android.app.ActivityThread.-wrap11(Unknown Source:0) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1589) 
        at android.os.Handler.dispatchMessage(Handler.java:106) 
        at android.os.Looper.loop(Looper.java:164) 
        at android.app.ActivityThread.main(ActivityThread.java:6494) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807) 
```

dunderhay, 2018-07-07:
ah okay DeleteAllCardsPreference was importing `android.preference.DialogPreference;` instead of `android.support.v7.preference.DialogPreference;`

dunderhay, 2018-07-07:
This should now be resolved in commit [a3ce7f9d9acf38a0b76665a2d5335af015a54991](https://github.com/TeamWalrus/Walrus/commit/a3ce7f9d9acf38a0b76665a2d5335af015a54991) 🤞

megabug, 2018-09-08:
Fixed by aaa1f09.

closed 2018-09-08 by megabug