TeamWalrus/Walrus #99: ChameleonMini: emulate different card types or maybe use CLONE

gtpy, 2018-06-21:
https://github.com/TeamWalrus/Walrus/blob/cbaf078278ed2ca6dfff1d61672076d09373caf7/app/src/main/java/com/bugfuzz/android/projectwalrus/device/chameleonmini/ChameleonMiniDevice.java#L321
https://github.com/TeamWalrus/Walrus/blob/cbaf078278ed2ca6dfff1d61672076d09373caf7/app/src/main/java/com/bugfuzz/android/projectwalrus/device/chameleonmini/ChameleonMiniDevice.java#L183

you configure the slot to #iso14443a_reader” and then use the „identify“ command, but slot will get only config=Mf_classic_1k every time.
Maybe it would be easier to use the “clone” command (which switches to reader mode! Identify and set the slot to identified configuration and clone the uid.

Maybe I have overseen something, hadn’t yet enough time to read the whole coding.

dunderhay, 2018-08-04:
We have had some discussion around the issue and wanted to use the device more closely to it's original use case. We feel that the command usage for Chameleon mini should be as follows:

| UI Button  | Functionality |
|------------|:--------------:|
|  Write | Write to a Chameleon mini card slot |
|  Read  | Read from a Chameleon mini card slot |
|  Emulate | Not supported by device |

The process is:
1. Use a Proxmark3 to obtain a Mifare card dump and save that to your Walrus wallet. 
2. Upload the card from your Wallet to the Chameleon Mini.
3. Use Chameleon mini without being attached to anything. 

At some stage we will also add the ability to import Mifare card dumps via Walrus. 

I think I am making the assumption that no card access system uses Mifare UID only, and so using the CLONE command to copy the UID only seems kinda pointless? I dunno, open to suggestions and discussion around this.

dunderhay, 2019-03-10:
The `CLONE` command copies the data directly to a Chameleon mini card slot, the `IDENTIFY` command reads the card uid, atqa, and sak that can be used to create and store the data in the Walrus wallet. This can then be used later to write the data to a card slot.

closed 2019-03-10 by dunderhay