TeamWalrus/Walrus #101: Support Chameleon Mini Rev.E Rebooted

dunderhay, 2018-07-27:
https://lab401.com/products/chameleon-mini-reve-rebooted
https://github.com/iceman1001/ChameleonMini-rebooted

phpsystems, 2018-12-03:
+1 for this request.

peseta3, 2018-12-18:
Hi Together,

I get a ChameleonMini RevE Rebooted and install the Walrus Ap on my Mobilephone Galaxy A3 (2017) and I didn´t get e conection.
The Device is not to see in the App.
What do I false?
Please help me.

gtpy, 2018-12-18:
Hi,
are you using an OTG Adapter?

-------- Original-Nachricht --------

peseta3, 2018-12-18:
Hi
Yes sure..

peseta3, 2018-12-18:
by the way prokmark3 v2 is conected.

iceman1001, 2019-02-23:
Heads up,   the latest firmware version is starting to use different name,  if you have that as base for identification of device.

nicoschtein, 2019-05-17:
As of 0d5d7e918d460a1f03bb2fe41c64a6827bce47b9 built from source, RevE Rebooted with latest Iceman FW is detected but i don't have any options under "Choose card type". Is this expected?

dunderhay, 2019-06-29:
Hey @nicoschtein,

RevE Rebooted doesn't really support "reading" cards as on RevG. At the moment I don't think the Chameleon mini is really meant to be used for reading cards really, more for acting like a hardware wallet from my understanding.