TeamWalrus/Walrus #102: Support Proxmark3 RDV4.0

dunderhay, 2018-07-27:
http://proxgrind.com/prototyping/proxmark3-rdv4-0-development/

MeulemanSven, 2018-07-27:
I have a proxmark3 rdv4, if you want I can test beta's for you. At the moment I only have Mifare 1k cards to play with but I can try to get others too. The release version of walrus recognizes the proxmark but the dans-branch and the master branch not at this moment (See my remark on issue #97)

dunderhay, 2018-07-27:
Hey @MeulemanSven,

Thanks for the response. I had some feed back that the proxmark3 rdv4 cannot be powered via the phone as it requires higher voltage, unsure as have not tested myself yet. It would need an external battery pack if that was the case. I will also check out the vendor-ids changing as you mentioned in in your remark on issue  #97

netvader, 2018-07-31:
I've tested Proxmark3 RDV4.0 with Nexus 5x (Oreo 8.1) and a short [Amazon Basic Cable](https://www.amazon.com/dp/B01LONQBDG) together with Walrus from Play Store and I also compiled  the matts-branch. Both Versions found my Proxmark3 RDV4.0 and with the DEBUG Version I could read the Mifare Classic 1K UID. I guess it depends on the smartphone USB/OTG power?

akmalhisyam, 2018-07-31:
I've tested rdv4.0 with my Xiaomi Mi A1 and Walrus able to detect it. But to make it able to read cards, I had to flashed the official firmware. Iceman refused to work.

original13, 2018-08-13:
Any luck getting this working with iceman?

dunderhay, 2018-08-25:
@original13 yeah however requires a small tweak to iceman fork in lfdemod operation. will make a pull request soon

original13, 2018-08-28:
Thanks Dunderhay... Really looking forward to it. Your app is "A"wesome

original13, 2018-08-30:
Any Update on ETA ? Anything we can do to help?
Cheers!

dunderhay, 2018-08-30:
@original13 will do a pull request to iceman fork this evening or tomorrow afternoon. Sorry for the delays!

original13, 2018-09-01:
Any luck? Is this a change required in Iceman firmware or your walrus app?

dunderhay, 2018-09-02:
@original13 the change is required in iceman firmware. We have made a fork with changes, need to do a pull request :) @megabug is dealing with this, and appears there are two slightly different issues. More details coming! Sorry we only get our weekends to do this!

original13, 2018-09-05:
Cool... Eagerly awaiting the update 👍

iceman1001, 2018-09-05:
No need to wait. I push your fixes already and others already.

original13, 2018-09-05:
Awesome! Thanks!

dunderhay, 2018-09-08:
As per issue #108, iceman1001's fork has been fixed and we have successfully tested the RDV4 on our device without requiring an additional power source during the card read. If you still can't get this working and have verified this is not a power issue (i.e. by using a terminal app to get a successful read), please feel free to open another issue.

closed 2018-09-08 by dunderhay

Cheeks300zx, 2018-11-21:
I have a AT&T Samsung Galaxy Note 9 500GB and it works through a Amazon Basic cable 1 feet and 6 inch version to power my Proxmark RDV4.0 with no problem. It can read and write HID cards. I also use it with my &Prox app which works with all the cards.