TeamWalrus/Walrus #112: Unable to scan cards with current iceman fork

cyb3rwr3ck, 2018-12-17:
Fresh build and install to my PM3 RDV2:
 [ ARM ]
 bootrom: iceman/master/ice_v3.1.0-864-g99d7546c 2018-12-17 17:57:01
 os: iceman/master/ice_v3.1.0-864-g99d7546c 2018-12-17 17:57:03
 [ FPGA ]
 LF image built for 2s30vq100 on 2017/10/25 at 19:50:50
 HF image built for 2s30vq100 on 2017/11/10 at 19:24:16

The App is working as expected, PM3 is recognized and the tuning for rf/lf is working but no cards are recognized at all via the scanning functions. Did a cross check via the cli client and it is working as expected so no general issue with the PM3.

dunderhay, 2019-03-10:
Hi @cyb3rwr3ck,

We just just pushed out version 0.2.1 of Walrus. Could you please try again?

cyb3rwr3ck, 2019-03-10:
I can confirm that its working with os ice_v3.1.0-1075-ga5dfd9b1 and both mifare and hid cards now. 

What was kind of surprising to me is the fact that it is not possible to emulate the captured cards (hid or mifare) using the proxmark. It just returns a message stating that "no connected device can emulate this kind of card" although the proxmark is capable of doing so.

dunderhay, 2019-03-11:
Great - sounds like we are making progress - I will test the emulate feature you have mentioned and we can open a new issue :)

closed 2019-06-29 by dunderhay