TeamWalrus/Walrus #113: RDV4 power up by smartphone USB/OTG (IOT use Walrus)

Mera-balou, 2019-02-09:
This issue is not directly linked to walrus, but i don't manage to power up my PM3 RDV4 with my phone (in order to use it with walrus app). I've an One Plus 6T smartphone.
@netvader suggested [here](https://github.com/TeamWalrus/Walrus/issues/102#issuecomment-409184051)   that it depends on the smartphone USB/OTG power.
-  Well, did anyone succeed in powering up RDV4 with a smartphone?
- If so, with wich smartphone/cable models?
- Anyway 

to increase my OP 6T OTG power?

Regards

iceman1001, 2019-02-13:
Normally,   a y-cable usb and external usb battery.  One goes to phone, one goes to pm3, one goes to usb-battery.

kimocoder, 2019-02-13:
USB OTG power and extra support may be enabled in a custom kernel.

iceman1001, 2019-02-13:
It seem that some smartphones are able to power the rdv4 direct without the need for usb battery.
I don't think we have a list over which.

Mera-balou, 2019-02-18:
Surprisingly, it worked with a smaller cable (amazon basic). But with your recommendation, i bought an Y cable to plug a power bank, IOT to increase antenna performance.

Thanx.

PS : final question for my own curiosity, last commit is from september. Any difficulties to maintain the project or you just wait bigger changes to commit?

Cheeks300zx, 2019-02-18:
I have AT&T Galaxy Note 9 that powers my RDV4.0 directly with no y-cable. I use a amazon basics cable. I power the proxmark with my phone with no problem. I use Walrus app and &Prox app with no issues.

closed 2019-03-01 by Mera-balou

ajroot1999, 2019-04-10:
I have a Galaxy S8 and am not having any luck getting the Proxmark3 V4 to read cards.  If I use a y cable to connect to both the phone and a USB battery will that generate enough power for the antenna to read?

iceman1001, 2019-04-10:
first of all,  have you compiled/flashed lastest firmware  (RRG repo) for your RDV4?