TeamWalrus/Walrus #122: Feature Request

sho-luv, 2019-05-31:
First off let me say I love this project. I talk it up to everybody who might be in the space to use it. I have used it on multiple projects of my own. Including one where I found all the badges of a company on a document which I manually entered into Walrus and was able to write multiple useful cards. Thanks ;)

I wanted to put in a feature request. If I could import projects or cards and export them. I would find that useful. I just sent screenshots or manually text them currently. However, if I could just export them in whatever format and import them. That would be extremely useful in the field when working with multiple people.

dunderhay, 2019-06-13:
Hey @sho-luv,
Thanks for the positive feed back and glad you are enjoying the app. I think this is a great feature request so we can put it in the que to get done at some point :P  
Cheers!

dunderhay, 2019-06-29:
Just noticed this is a duplicate of #21