TeamWalrus/Walrus #124: Proxmark3 RDV4.0 w/standalone module stuck in idle

enporter, 2019-08-26:
So for starters I have: 
Proxmark3 RDV4.0 w/standalone BT with battery
Unitek adaptor cable
Xiaomi Redmi 7a
Walrus (from Playstore)
RDV4.0 firmware specs in the screenshot (using latest from iceman)
[image]

So it connects and is recognized no problem but it won't read cards AND it fails to show version information AND tuning LF or HF is futile as it fails. 

I thought maybe a power issue so I have an Orico powered USB that was actually used in another issue on here but it didn't work. 
I tried using it with the battery on. Didn't help. 
I tried it without the add-on and just used original hardware. Didn't work. 
I tried swapping adapters. Didn't work (I use them for other devices so I know they work anyways)

I looked around and saw some iceman for and power issues but I don't think I saw an issue with the proxmark3 being detected and just sitting an idle. If this has already been reported just let me know and close it. 

Any ideas would be appreciated.

evazzoler, 2020-01-15:
I have the same issue. No reply since months, is this project abandoned?

enporter, 2020-01-15:
If I remember correctly the team emailed me and said the proxmark3's have to be rolled back to earlier firmware and they are waiting for the RRG Fork to get more developed before doing anything else. That was awhile ago though. I wanted to keep the additional functionality so I just use RRG rfid app which has the command line client. Patiently waiting as I think Walrus has really cool potential.

joetomasone, 2021-06-23:
Definitely appears to not support RRG/IceMan.   Will definitely need that support to be usable by many Proxmark aficionados..

enporter, 2021-06-23:
> Definitely appears to not support RRG/IceMan. Will definitely need that support to be usable by many Proxmark aficionados..

I think it's a unicorn 😥 You hope it's real one day but maybe it only exists on Github history 😥😥😥

I wish I was knowledgeable enough to take it on