TeamWalrus/Walrus #125: Домофоны

Sanprox, 2020-09-18:
Если добавить эти теги распознавание домофонных ключей то получится топчик
[image]
![Screenshot_20200916_004928_ikey 

> ikeybase](https://user-images.githubusercontent.com/71011794/93578109-96043680-f9a5-11ea-97ce-64cd59ee4bd0.jpg)

Sanprox, 2020-09-18:
Брелки (Техком) proxmark3 Easy не читает вообще работают на 13.56. Купил proxmark думал аппарат супер но он даже не смог домофонный ключ скопировать...

assasinfil, 2021-01-26:
> Брелки (Техком) proxmark3 Easy не читает вообще работают на 13.56. Купил proxmark думал аппарат супер но он даже не смог домофонный ключ скопировать...

Просто кто-то не умеет пользоваться)
Если речь идет про t55xx, то я сталкивался с такой же проблемой.
Там нужно вручную продублировать сектора 1 2 в второй блок иначе домофон палит клон.